cjhdev/lora_device_lib #2: project name is technically in violation of LoRa Alliance guidelines

cjhdev, 2019-11-07:
The LoRa Alliance advises against incorporating 'lora' or 'lorawan' into the name of a product.

[https://lora-alliance.org/resource-hub/lora-alliancer-marks-logo-usage-policy-and-guidelines](https://lora-alliance.org/resource-hub/lora-alliancer-marks-logo-usage-policy-and-guidelines)

This was brought to my attention when the loraserver project changed its name to chirpstack. 
This project will probably do the same.

C47D, 2019-11-19:
I liked the LDL prefix on the API, are you still looking for a new project name?

cjhdev, 2019-11-19:
I'm also OK with the prefix. I updated the project to use it everywhere for consistency.

I don't have any ideas regarding name change. Even if the name changes I won't bother changing the prefix.