cjhdev/lora_device_lib #3: How often should we call LDL_MAC_process?

C47D, 2019-11-11:
Hi,

I am wondering if we should call `LDL_MAC_process` at least X times per second?

Regards

cjhdev, 2019-11-11:
There is no fixed rate, instead `LDL_MAC_process` must be called when `LDL_MAC_ticksUntilNextEvent` returns 0.

In practice if you have a mainloop and you aren't sleeping, it is sufficient to poll `LDL_MAC_process` once per loop iteration without ever checking `LDL_MAC_ticksUntilNextEvent`.

C47D, 2019-11-11:
Thanks for the reply, in case of my MCU being in low power mode (sleeping), I guess I can do something like below?

```c
while (1) {
    sleep_ticks = LDL_MAC_ticksUntilNextEvent();
    sleep_mcu(sleep_ticks) // sleeps the MCU for x ticks

    LDL_MAC_process();
}
```

cjhdev, 2019-11-12:
Yes that will work. `LDL_MAC_ticksUntilNextEvent` exists for this very purpose, to let you know how long you can sleep for. 

It goes without saying that your time source needs to keep working in sleep. On my STM32L port I use LPTIM.

C47D, 2019-11-12:
You're right, I totally forgot about the low power peripherals on the STM32L chips, I am using a STM32L476 and was thinking on use a not very low power sleep mode to keep the systick running.

cjhdev, 2019-11-12:
A limitation of LDL at the moment is that it expects to always be running. If you stop and re-start the time source, you have to update the result of LDL_System_time() (i.e. move it forward) but you can only do that so much before the internal timers wrap around.

A future feature might be some kind of suspend and resume feature to safely take care of all of this.

C47D, 2019-11-12:
Does loramac-node implement something along this lines? I haven't seen anything like this on their examples, maybe I should check the issues and see if somebody have asked about low power modes.

cjhdev, 2019-11-13:
I don't think so, suspending/resuming as described is a bit niche. They have a start and stop interface but it doesn't seem so useful.

closed 2021-03-03 by cjhdev