cjhdev/lora_device_lib #4: SX1262 support

RX14, 2019-11-18:
You mentioned in another issue you'd like some kind of coordination before others submitted PRs. I have some [SX1262](http://www.dorji.com/products-detail.php?ProId=65) modules arriving soon, and I'd like to add support to this library (I can see where, and how). Just wanted to hear your thoughts on contributing this upstream before I start.

I share your frustration with the (over)complexity of the other implementations, after looking at either using semtech's reference which has SX1262 support already, or adapting arduino-lmic.

cjhdev, 2019-11-18:
The new radios look great, I would be very happy to accept code to support them.

For contributing a radio driver I think the guidelines to observe are:

- must be contributed with the same license (it's still your code, of course)
- must have an opt-in build option (e.g. LDL_ENABLE_SX1262)
- must be selectable at run-time (like the existing radio drivers)
- must keep consistent style (mainly concerned about indentation and spaces)

Hopefully the run-time selection feature doesn't cause too many problems. It was easy with the SX1272/6 since it's almost the same chip. I like it because it will make it possible have universal wrappers on Arduino style platforms that have plenty of flash.

I am interested to see if the existing Mac->Radio interfaces make sense for a new type of radio. They can be modified if necessary.

I'm not currently using any well thought out branching strategy. Assume that I will accept a pull-request to master if it passes review. If people start depending on this library I might shift to something like gitflow to stage releases.

Word of warning, I'm about to commit a bunch of interface changes to master. A lot of it is silly stuff like making the prefixes consistent (no more mixed lora/ldl). There is a slight change to how the Radio interrupt lines connect back to the MAC. Once this is done I hope it can remain stable for a long time.

RX14, 2019-11-18:
The guidelines are sane, and what I would have done anyway :)

> Arduino style platforms that have plenty of flash

I'm gonna be using an STM32l431, so not a problem for me!

> I'm not currently using any well thought out branching strategy. Assume that I will accept a pull-request to master if it passes review. If people start depending on this library I might shift to something like gitflow to stage releases.

Just a side note, but I personally don't really like gitflow's `develop` and `master` split - releases should use git tags and development should happen on `master`.

> Word of warning, I'm about to commit a bunch of interface changes to master. A lot of it is silly stuff like making the prefixes consistent (no more mixed lora/ldl). There is a slight change to how [...]

Thanks for the heads up! I'm not likely to start working on this until the module works it's way to me from China, and even then I'm quite busy working on a cubesat, so this PR may never materialize.

cjhdev, 2019-11-19:
> I'm gonna be using an STM32l431, so not a problem for me! I'll test the module on arduino before possibly porting the arduino wrapper to STM32duino.

I like Arduino wrappers. Aside from being easy to use, they are a good chance to sanity check the interfaces and portability features. I'd like to have more of them. 

> Just a side note, but I personally don't really like gitflow's develop and master split

I've actually never used it. A policy of "use a tagged release" is much simpler.

> PR may never materialize.

No worries.

cjhdev, 2021-01-18:
0.5.0 supports SX1262 :tada:

closed 2021-01-18 by cjhdev