cjhdev/lora_device_lib #5: OTAA join

dzurikmiroslav, 2020-01-25:
Hi, I am glad to see there is new alternative to LMIC or LoRaMac-Node for no-arduino developers.
I am using ESP32, in my code https://github.com/dzurikmiroslav/esp32-ldl-example I have issue with OTAA joining request. In log I see many attempts to join, but in TTN console I see only first attempt. After restart is the situation same, in TTN console will be added one join request.

Application log: 
```
1229: LDL_MAC_addChannel: debug: adding chIndex=0 freq=868100000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=1 freq=868300000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=2 freq=868500000 minRate=0 maxRate=5
I (318) main: LDL_MAC_RESET
I (318) main: LDL_MAC_STARTUP
216: LDL_MAC_otaa: debug: sending join in 16359821 ticks
I (76518) main: LDL_MAC_TX_BEGIN
I (76618) main: LDL_MAC_TX_COMPLETE
I (81618) main: LDL_MAC_RX1_SLOT
I (82518) main: LDL_MAC_RX2_SLOT
I (83318) main: LDL_MAC_JOIN_TIMEOUT
950: LDL_MAC_process: debug: dither retry by 18069775 ticks
I (107518) main: LDL_MAC_TX_BEGIN
I (107718) main: LDL_MAC_TX_COMPLETE
I (112618) main: LDL_MAC_RX1_SLOT
I (113618) main: LDL_MAC_RX2_SLOT
I (114618) main: LDL_MAC_JOIN_TIMEOUT
950: LDL_MAC_process: debug: dither retry by 4623390 ticks
I (130018) main: LDL_MAC_TX_BEGIN
I (130318) main: LDL_MAC_TX_COMPLETE
I (135218) main: LDL_MAC_RX1_SLOT
I (136218) main: LDL_MAC_RX2_SLOT
I (137218) main: LDL_MAC_JOIN_TIMEOUT
950: LDL_MAC_process: debug: dither retry by 18470742 ticks
I (174118) main: LDL_MAC_TX_BEGIN
I (174618) main: LDL_MAC_TX_COMPLETE
I (179518) main: LDL_MAC_RX1_SLOT
I (180518) main: LDL_MAC_RX2_SLOT
I (181518) main: LDL_MAC_JOIN_TIMEOUT
950: LDL_MAC_process: debug: dither retry by 17685113 ticks
I (236118) main: LDL_MAC_TX_BEGIN
I (237118) main: LDL_MAC_TX_COMPLETE
I (242018) main: LDL_MAC_RX1_SLOT
I (243018) main: LDL_MAC_RX2_SLOT
I (244718) main: LDL_MAC_JOIN_TIMEOUT
```
TTN console:
[image]

cjhdev, 2020-01-25:
Hi dzurikmiroslav

I just tried master on my avr demo board to see if I've broken anything lately. I've been resetting the node to see if I can get the join to fail but for me it works correctly nearly every time. 

I managed to reproduce your problem once but wasn't able to collect the frame log, and now it continues to work perfectly.

I suspect there might be some edge case where the cached join request message is clobbered. I want to confirm that the exact same join accept message gets sent on each retry. Can you see the gateway traffic log?

If you don't have access, don't worry, I'll set something up on my simulator but it might take me a few days.

dzurikmiroslav, 2020-01-25:
I were try instead of ESP-IDF use Arduino framework, and code start working normally. Problem were that in `LDL_Chip_write` and `LDL_Chip_read` I used `spi_device_transmit`, which make SPI transaction. Arduino `SPI.transfer` not make new transaction...

dzurikmiroslav, 2020-01-27:
After more testing, I reached the problem is when first join attempt is unsuccessful (maybe signal strength problem etc...). When first join attempt is successful, node start sending data.

I modify handler function:
```
        case LDL_MAC_TX_BEGIN:
            ESP_LOGI(TAG, "LDL_MAC_TX_BEGIN devNonce=%d", mac.devNonce);
            break;
```

Log from successful join:
```
I (94) main: Starting...
1229: LDL_MAC_addChannel: debug: adding chIndex=0 freq=868100000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=1 freq=868300000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=2 freq=868500000 minRate=0 maxRate=5
I (94) main: LDL_MAC_RESET
I (94) main: LDL_MAC_STARTUP
I (59994) main: LDL_MAC_ready
I (59994) main: LDL_MAC_otaa
216: LDL_MAC_otaa: debug: sending join in 23758066 ticks
I (83694) main: LDL_MAC_TX_BEGIN devNonce=23989
I (83794) main: LDL_MAC_TX_COMPLETE
I (88694) main: LDL_MAC_RX1_SLOT
I (88794) main: LDL_MAC_DOWNSTREAM
1229: LDL_MAC_addChannel: debug: adding chIndex=0 freq=868100000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=1 freq=868300000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=2 freq=868500000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=3 freq=867100000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=4 freq=867300000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=5 freq=867500000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=6 freq=867700000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=7 freq=867900000 minRate=0 maxRate=5
I (88894) main: LDL_MAC_JOIN_COMPLETE
I (89494) main: LDL_MAC_ready
I (89494) main: LDL_MAC_unconfirmedData
I (89494) main: LDL_MAC_TX_BEGIN devNonce=23990
I (89494) main: LDL_MAC_TX_COMPLETE
I (90494) main: LDL_MAC_RX1_SLOT
I (91494) main: LDL_MAC_RX2_SLOT
I (91894) main: LDL_MAC_DATA_COMPLETE
```

Log from unsuccessful join:
```
I (94) main: Starting...
1229: LDL_MAC_addChannel: debug: adding chIndex=0 freq=868100000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=1 freq=868300000 minRate=0 maxRate=5
1229: LDL_MAC_addChannel: debug: adding chIndex=2 freq=868500000 minRate=0 maxRate=5
I (94) main: LDL_MAC_RESET
I (94) main: LDL_MAC_STARTUP
I (59994) main: LDL_MAC_ready
I (59994) main: LDL_MAC_otaa
216: LDL_MAC_otaa: debug: sending join in 39518844 ticks
I (99494) main: LDL_MAC_TX_BEGIN devNonce=12023
I (99494) main: LDL_MAC_TX_COMPLETE
I (104494) main: LDL_MAC_RX1_SLOT
I (105494) main: LDL_MAC_RX2_SLOT
I (106194) main: LDL_MAC_JOIN_TIMEOUT
950: LDL_MAC_process: debug: dither retry by 17082309 ticks
I (129394) main: LDL_MAC_TX_BEGIN devNonce=12023
I (129494) main: LDL_MAC_TX_COMPLETE
I (134494) main: LDL_MAC_RX1_SLOT
I (135494) main: LDL_MAC_RX2_SLOT
I (136494) main: LDL_MAC_JOIN_TIMEOUT
950: LDL_MAC_process: debug: dither retry by 27441593 ticks
I (174694) main: LDL_MAC_TX_BEGIN devNonce=12023
I (174894) main: LDL_MAC_TX_COMPLETE
I (179894) main: LDL_MAC_RX1_SLOT
I (180894) main: LDL_MAC_RX2_SLOT
I (181794) main: LDL_MAC_JOIN_TIMEOUT
950: LDL_MAC_process: debug: dither retry by 29505280 ticks
I (229794) main: LDL_MAC_TX_BEGIN devNonce=12023
I (230094) main: LDL_MAC_TX_COMPLETE
I (235094) main: LDL_MAC_RX1_SLOT
I (236094) main: LDL_MAC_RX2_SLOT
I (236994) main: LDL_MAC_JOIN_TIMEOUT
```

After restart `devNonce` is randomly generated, but after unsuccessful join request still same. Maybe problem ist that the `devNonce` is randomly generated in `LDL_MAC_otaa`, but 
 `LDL_MAC_otaa` is called once...

cjhdev, 2020-01-27:
Yes, the dev nonce only changes when you initiate otaa, not on a retry.

TTN (the community service at least) will drop a dev nonce that it has seen before and the node will keep sending retries for ever. That's what it looks like when it happens to me.

The 1.0 spec says to use a random dev nonce but collisions happen since it's a 16 bit number. You can ensure this doesn't happen by not defining LDL_DISABLE_POINTONE and allowing LDL to operate in 1.1 mode (which is backwards compatible to 1.0). The downside of this mode is that you will need to save/restore dev nonce or else you run into the same problem.

Long story short, the random dev nonce is good for experiments since you can just reset to clear the block, but 1.1 is superior for nodes you put in the field.

cjhdev, 2020-09-30:
I'm closing this issue since it seems to be dead.

I wasn't able to reproduce the join timeout, it may have been timing related. The devNonce will not change until OTAA succeeds.

closed 2020-09-30 by cjhdev