cjhdev/lora_device_lib #9: Join timeout

dzurikmiroslav, 2020-03-20:
Hi,

I am build my own single channel gateway using ESP-1ch-Gateway and connect it to TTN. In TTN console I see received join query and joint accept response. My device always give me LDL_MAC_JOIN_TIMEOUT. 

This problem may be caused by wrong LDL_System_advance value? In what way I should calculate it?

cjhdev, 2020-03-20:
Hi

Single channel gateways are a problem for many reasons. I am surprised you are able to get a join request to the server.

Setting LDL_System_eps() is usually enough to ensure you will hit the RX window. LDL_System_advance() simply ensures the window opens n ticks earlier than it otherwise would. 

Events LDL_MAC_RX1_SLOT and LDL_MAC_RX2_SLOT will indicate a missed window if rx_slot.error > rx_slot.margin.

Other potential problems:

1. TTN might not be delivering the downlink at the right time
2. If you are using SF12, it doesn't work reliably unless you have a temperature compensated oscillator (This is why LDL_DEFAULT_RATE defaults to 1)
3. The radios might be too close together

cjhdev, 2020-09-29:
I'm closing this issue because I've not been able to reproduce the issue.

The error compensation method has been updated recently to make it easier to apply "fudge factor" to try to get around window timing problems. I've written a section in porting.md about it.

closed 2020-09-29 by cjhdev