cjhdev/lora_device_lib #10: OTAA not recieving JoinAccept

abdouTCH13, 2020-08-04:
Hello ,
i m having some troubles connecting my node to TTN as the node is stuck in the join procedure.
i thought it was a timer issue ,i was using stm32 1ms systick timer as a counter then i switched to lptim which is ticking at a rate of 16384 tick per second but the problem still persist i m still not recieving the joinaccept packets.
I've set a breakpoint in LDL_MAC_process() where LDL_Radio_collect() is called but its not reaching that point.

I'm using STM32L073RZ microcontroller with the semtech sx1272 transceiver , laird RX186 gateway and the TTN server stack
my node is about five meters away from the gateway 
the joinrequest messages are received by TTN and its sending the join accept messages back.

also i tried to enlarge RX1/RX2 windows but wont work either 
[https://github.com/cjhdev/lora_device_lib/blob/94c64ba1cae06d81ee4d3b80d4895b543284ec27/src/ldl_mac.c#L570
](url)
[https://github.com/cjhdev/lora_device_lib/blob/94c64ba1cae06d81ee4d3b80d4895b543284ec27/src/ldl_mac.c#L616
](url)

here is the link to my test repo
[https://github.com/abdouTCH13/LDL_STM32L073RZ_test](url)

any help would be very appreciated.
thank you

cjhdev, 2020-08-05:
Hi

I will try to reproduce this problem and get back to you. Might take a few days.

abdouTCH13, 2020-08-08:
hello i have resolved the issue after installing the chirpstack servers localy
it was my mistake i misscalculated the eps value and now everything is working properly
thank you for your help .

closed 2020-08-08 by abdouTCH13