cjhdev/lora_device_lib #15: Proposal to remove diagnostic events from ldl_mac_response_type

cjhdev, 2020-10-04:
I want to remove the diagnostic events because this information isn't useful to applications and yet takes code space and increases the line count. Previously these events could be enabled/disabled at compile time but I removed the option as it made the configuration documentation more complicated.

Since the information is sometimes useful to eyeball during development, I propose that the information should be printed via LDL_DEBUG. This is typically what these events get used for anyway, at least in the ports I've written.

Diagnostic events include:

- LDL_MAC_RX1_SLOT
- LDL_MAC_RX2_SLOT
- LDL_MAC_DOWNSTREAM
- LDL_MAC_TX_COMPLETE
- LDL_MAC_TX_BEGIN

frbehrens, 2020-10-04:
I'm fine with that, if it saves some flash memory. Please let it available with DEBUG macros, some of the events are very usefull during application development.

cjhdev, 2020-10-28:
Completed in 0.4.5.

closed 2020-10-28 by cjhdev