cjhdev/lora_device_lib #16: Proposal to refactor LDL_DEBUG, LDL_ERROR, and LDL_INFO

cjhdev, 2020-10-04:
These macros currently look like this:

`LDL_DEBUG(APP, ...)`

The APP argument is a hangover from some time ago when I thought it was useful to be able to link a message back to some instance. In practice a suitable pointer isn't always available, and even if one is, there is usually only on instance producing messages and so its redundant.

The `...` usually expands to something like:

`LDL_DEBUG(NULL, "a format string with some number %d", 42)`

This is a problem for anyone using AVR since you have no option of putting the format string into progmem. 

I propose changing the pattern to:

`LDL_DEBUG(FMT, ...)`

This could then be redefined on your port as:

`LDL_DEBUG(FMT, ...) printf(FMT, __VA_ARGS__);`

Or if you are on AVR and want the string in flash:

`LDL_DEBUG(FMT, ...) printf(PSTR(FMT), __VA_ARGS__);`

etc.

frbehrens, 2020-10-05:
1. __I support your proposal__. The APP argument makes no sense currently.
2. It was already possible to store the format string into progmem. The compiler is able to remove the unused APP argument. My definition is
```c
extern void LDL_DBGINF(const char *__fmt, ...);
#define LDL_DEBUG(APP,FMT, ...) LDL_DBGINF(PSTR(FMT), ##__VA_ARGS__);
```

3. I don't like the last refactoring, that inserts the \__FUNCTION\__ argument. It adds more program code, requires more flash memory and I don't see any advantage in debug output. I did not merge these changes into my local branch.

cjhdev, 2020-10-15:
Oh yes, you're right. I've confused myself :smile:

I've gone ahead and removed the app argument and function macros.

cjhdev, 2020-10-28:
Completed in 0.4.5.

closed 2020-10-28 by cjhdev