cjhdev/lora_device_lib #25: preferred way to use DeviceTimeSync

szemzoa, 2021-02-26:
Hi,

How do you use the DeviceTimeSync options with the current implementation? 

I've added a new app callback function to the end of the processTX function, where I only record the current msec/tick counter, and calculate the time diff at the LDL_MAC_DEVICE_TIME callback, but I'm not sure this is the best way.

cjhdev, 2021-02-27:
Hello

I have a solution that implements the algorithm you describe, but integrated without extra callbacks. I'll merge this in.

The L2 spec (1.0.4) doesn't actually say when the server should record a timestamp. Looks like it should be the time at TX complete (from device perspective) but I only find this out by looking at TTN and ChirpStack.

cjhdev, 2021-02-28:
The changes are now in master. 

I'm yet to try it on real hardware.

szemzoa, 2021-03-02:
Thanks for the update, I'll test it in a few days.

szemzoa, 2021-03-03:
The changes are working for me with an accuracy of about ~20msec, so I close the issue, thanks!

closed 2021-03-03 by szemzoa