cjhdev/lora_device_lib #29: Add a check for maximum message size on send

frbehrens, 2021-02-28:
There is no explizit check for maximum message size. If the size is too big, it leads to the (wrong) error message LDL_STATUS_MACPRIORITY / "mac data has been prioritised".

I propose an explicit check like mentioned in the patch. I know, that the style does noct match the library, therefore I did not create a pull request.
```C
--- "a/src/ldl_mac.c"
+++ "b/src/ldl_mac.c"
@@ -1656,40 +1665,45 @@ static enum ldl_mac_status externalDataCommand(struct ldl_mac *self, bool confir
     if(self->ctx.joined){
 
         if(self->op == LDL_OP_NONE){
 
             if((port > 0U) && (port <= 223U)){
 
                 if(self->band[LDL_BAND_GLOBAL] == 0U){
 
                     /* set desired power and rate */
                     self->tx.power = self->ctx.power;
 #ifdef LDL_DISABLE_TX_PARAM_SETUP
                     self->tx.rate = self->ctx.rate;
 #else
                     self->tx.rate = LDL_Region_applyUplinkDwell(self->ctx.region, uplinkDwell(self->ctx.tx_param_setup), self->ctx.rate);
 #endif
 
                     if(selectChannel(self, self->ctx.rate, 0U, &self->tx)){
 
                         LDL_Region_convertRate(self->ctx.region, self->ctx.rate, &sf, &bw, &maxPayload);
 
+                        if (len + LDL_Frame_dataOverhead() > maxPayload) {
+                            LDL_INFO("message is too large");
+                            return LDL_STATUS_SIZE;
+                        }
+
                         (void)memset(&self->opts, 0, sizeof(self->opts));
 
                         if(opts != NULL){
 
                             (void)memcpy(&self->opts, opts, sizeof(self->opts));
                         }
 
                         self->opts.nbTrans = self->opts.nbTrans & 0xfU;
 
                         {
                             self->trials = 0U;
 
                             (void)memset(&f, 0, sizeof(f));
 
                             self->op = confirmed ? LDL_OP_DATA_CONFIRMED : LDL_OP_DATA_UNCONFIRMED;
 
                             f.type = confirmed ? FRAME_TYPE_DATA_CONFIRMED_UP : FRAME_TYPE_DATA_UNCONFIRMED_UP;
                             f.devAddr = self->ctx.devAddr;
                             f.counter = U16(self->ctx.up);
                             f.adr = self->ctx.adr;
```

cjhdev, 2021-02-28:
Thanks for pointing this out. I've pushed a fix based on your patch to master.

frbehrens, 2021-02-28:
Thanks!
I pulled your changes.

closed 2021-02-28 by frbehrens