cjhdev/lora_device_lib #30: Increment devNonce on every join request

frbehrens, 2021-02-28:
I observed sometimes problems with joins, where a reduced rate is required due to a longer distance. I debugged this further and could see, that the TTN ignores repeated join frames with the same devNonce. That happens, when a join-accept downlink is lost, but not when the join-request uplink is lost.

The LoRaWAN standard (1.0.4) seems to require a different devNonce as well: "For each transmission of a Join-Request, the end-device SHALL increment the DevNonce value."

So I propose to increment the devNonce in LDL for **every** join request frame. 

I created a PoC (https://github.com/frbehrens/lora_device_lib/commit/8e7c8acd76036a16cd3c07b6faa060e44930b24a), that solved the problem and allows a stable device join procedure.

What do you think about the issue?

cjhdev, 2021-03-01:
I must confess I haven't seen this problem but then I don't have much experience with devices in the field.

It doesn't make sense to me that the server would mark a join request as already seen if it hasn't received confirmation that the device that it received the join accept. The confirmation being a data frame encrypted with the new session keys.

I looked up that line in the L2 spec. They talk about "Join Request" as well as "Join Request frames". Are they the same thing?

I'll have a look at TTN and see if I can reproduce what you are seeing.

cjhdev, 2021-03-02:
You are right, the standard requires DevNonce to increment for each frame.

The "Technical Recommendations for Preventing State Synchronization Issues" lays out the reasons for why, and offers a workaround for the problem of running out of DevNonce increments.

szemzoa, 2021-03-02:
Hi,

It's not just an issue with TTN network, you are right, this is a problem on my Chirpstack based gw on my table, and I saw this Join problem with a Loriot server too. My very similar workaround works for me on both gateways.

cjhdev, 2021-03-02:
I've reproduced the problem in the integration test suite. This uses the Ruby wrapper and a headless ChirpStack instance.

I am expediting a fix based on frbehrens proof of concept.

frbehrens, 2021-03-02:
> I am expediting a fix based on frbehrens proof of concept.

Thanks! Please note, that I did not consider the random choice of devNonce.

frbehrens, 2021-03-07:
With https://github.com/cjhdev/lora_device_lib/commit/d443a9015fdc061fb710083e4dbe5f3936fae6b4 the problem is fixed.
Thanks! I'll close the issue.

closed 2021-03-07 by frbehrens