cjhdev/lora_device_lib #33: Not receiving Join Accept

marsam94, 2021-03-19:
Hello, 
I have 2 issues.
First of all I'm sometimes not getting any Join Accept from the TTN console for a few minutes, there are only Join Requests and nothing else. What can I do to fix that issue ?

And when I do get Join Accepts on the TTN console for some reason, I can't receive them. The ldl doesn't even get to the RX mode and instantly goes for a RESET. Maybe it has to do with the ticks configuration ? 

any help would be very appreciated.
thank you

marsam94, 2021-03-19:
The debug messages look like this :

> chIndex=0 freq=868100000 minRate=0 maxRate=5
chIndex=1 freq=868300000 minRate=0 maxRate=5
chIndex=2 freq=868500000 minRate=0 maxRate=5
set radio reset: ticks=39
clear radio reset: ticks=46
chIndex=0 freq=868100000 minRate=0 maxRate=5
chIndex=1 freq=868300000 minRate=0 maxRate=5
chIndex=2 freq=868500000 minRate=0 maxRate=5
OTAA is pending
add dither to otaa: ticks=111 delay=0
tx begin
ticks=162 freq=868500000 power=0  bw=125000 sf=7 size=23
offtime=100
interrupt fault
ticks=281
waiting to retry OTAA
radio fault detected, initiating radio reset
clear radio reset: ticks=318
channel is ready
add dither to otaa: ticks=6133 delay=0
tx begin
ticks=6187 freq=868300000 power=0  bw=125000 sf=8 size=23
offtime=100
interrupt fault
ticks=6397
waiting to retry OTAA
radio fault detected, initiating radio reset
clear radio reset: ticks=6434
channel is ready
add dither to otaa: ticks=16938 delay=0
tx begin
ticks=16990 freq=868100000 power=0  bw=125000 sf=9 size=23
offtime=100
interrupt fault
ticks=17367
waiting to retry OTAA
radio fault detected, initiating radio reset
clear radio reset: ticks=17405

Also I'm using the main loop given as an example

cjhdev, 2021-03-19:
Hi

Is LDL able to see a rising edge interrupt for DIO1?

The example skips over how to do this, the only mention that it is required is in [chip interface](https://ldl.readthedocs.io/en/latest/group__ldl__chip__interface.html) documentation.

marsam94, 2021-03-19:
You're right, I simply forgot to implement interrupts, it's working flawlessly now.
Thank you for your help

closed 2021-03-19 by marsam94