cjhdev/lora_device_lib #35: ticksUntilNextEvent keeps returning UINT32_MAX

marsam94, 2021-04-20:
Hello, 
I've been trying to use the library with my STM32L4 in Stop2 mode. 
I'm using a software 32bit timer with my LPTimer to make it work all together as you told me to in my last issue. 
The problem is happening at 2 different points in my program : 

- right after the boot state : any subsequent run in LDL_mac_process after some time resolves the issue
- in the joining process, during the wait_RX window : the debug messages say the channel is ready then I get UINT32_MAX and any subsequent run in LDL_mac_process doesn't help.

I'm using a SX1261 with the STM32L412KB, the timer used is LPTIM1 clocked at 1000KHz. The interrupt on pin DIO1 is getting handled and I'm calling LDL_Radio_handleInterrupt inside the callback. 

I actually had no issue when not going into stop mode and simply looping with the LDL_mac_process.

I've tried forcing the loop through LDL_mac_process using LDL_MAC_priority but then it means the mcu is looping during a whole TX-RX sequence.

Thank you for your help

cjhdev, 2021-04-22:
When LDL_MAC_ticksUntilNextEvent() returns UINT32_MAX it means that there are not any future events. It is not a bug.

The MBED wrapper source is a good worked example on how to port LDL. 

LDL::Device::worker() has a loop where LDL_MAC_ticksUntilNextEvent() is called right after LDL_MAC_process() to work out when a future event will occur. If there are no future events, the code doesn't set a timer.

Porting LDL in a way that plays well with stop mode is a lot of effort.

marsam94, 2021-04-23:
So it means that right after going through LDL_BOOT, there's no future event because we haven't started the otaa yet so that answers the first issue.

What I don't get is that it returns UINT32_MAX in the middle of the join process, right during the RX window so there should be future events at that moment. 

I really need to use lora with stop mode, so I guess I'll try using another library. 

Thanks for your help

closed 2021-04-24 by cjhdev